STMicroelectronics/x-cube-azrtos-h7 #1: Enabling TX_THREAD_ENABLE_PERFORMANCE_INFO causes Tx_Thread_Sync example freeze

vrabcak, 2020-12-22:
**Describe the set-up**

 * The board (either ST RPN reference or your custom board)

STM32H735G-DK 

 * IDE or at least the compiler and its version

STM32CubeIDE version 1.5.1

**Describe the bug**

If I uncomment `#define TX_THREAD_ENABLE_PERFORMANCE_INFO` in `tx_user.h`,  the Tx_Thread_Sync application freezes - the LEDs stop blinking.  
If I comment it out again, the app works fine.

ALABSTM, 2020-12-22:
Hi @vrabcak,

Thank you for this report. It will be forwarded to our development teams for deeper analysis. We will get back to you as soon as they provide their feedback. Thank you again.

With regards,

ALABSTM, 2020-12-24:
Hi @vrabcak,

Our development teams could reproduce the issue and analyse its root cause. They recommend to **clean and rebuild** the project **each time** the `tx_user.h` file is modified (**e.g.** after the `TX_THREAD_ENABLE_PERFORMANCE_INFO` is activated).

Below is their analysis:
* Observed behavior:
  *  Initially the project is built **without** activating the `TX_THREAD_ENABLE_PERFORMANCE_INFO` flag and the binary is correctly working.
  *  Then the `TX_THREAD_ENABLE_PERFORMANCE_INFO` flag is activated and the project is rebuilt **without cleaning**.
      * The issue **appears**.
      *  This is observed **only** in *DEBUG* mode.
* Analysis:
  *  When editing the `tx_user.h` file  and rebuilding the project the *IDE* will **only recompile** the files that are including that header file. But ThreadX comes with a set of `*.s` files that **do not include** the `tx_user.h`file. Thus they are **not recompiled**.
  *  The `TX_THREAD_ENABLE_PERFORMANCE_INFO` is **adding extra fields** to the `TX_THREAD` structure **increasing the overall size** of the structure.
      *  When rebuilding the project the `TX_THREAD` structures declared in the application will have a **greater size**.
      * They will occupy **extra room in memory** that was reserved for the objects generated from the `*.s` files that were **not recompiled**.
      * This yields **memory corruption**.
  *  The issue appears in *DEBUG* mode only because:
      * In *RELEASE* mode the compiler omits all unused symbols including the fields introduced by the `TX_THREAD_ENABLE_PERFORMANCE_INFO` flag.
      * While in *DEBUG* mode there is is **no optimization** at all.
* Recommendation:
  * The project should be cleaned and rebuilt **each time** the `tx_user.h` file is modified.

We hope this helps you solve the problem. Please let us know.

With regards,

ALABSTM, 2021-01-07:
Hi @vrabcak,

Any feedback from your side? Does our recommendation help you solve the issue?

With regards,

ALABSTM, 2021-01-11:
Hi @vrabcak,

Please allow me to close this issue. Thank you again for your contribution.

With regards,

closed 2021-01-11 by ALABSTM

ALABSTM, 2021-05-24:
ST Internal Reference: 98736