STMicroelectronics/x-cube-azrtos-h7 #2: newLib heap and ThreadX memory managment

bytesontherocks, 2021-01-15:
Hello, 

I have some experience with FreeRTOS, and the integration with the newLib heap and the heap mechanism for FreeRTOS was not fully integrated (see http://www.nadler.com/embedded/newlibAndFreeRTOS.html).

In the current ThreadX example I understand that the RAM map looks like that

 * ############################################################################
 * #  .data  #  .bss  #      _Min_Heap_Size +  _Min_Stack_Size    #   ThreadX memory Pool      
 * #            #          #                                                                   # 
 * ############################################################################
 * ^-- RAM start      ^-- _end                                                       ^ __RAM_segment_used_end__       _estack, RAM end --^

The newLib heap can expand to _estack (ORIGIN(RAM) + LENGTH(RAM)) - _Min_Stack_Size    according to the current "_sbrk" implementation that would mean that if you use malloc/new() at any point in your system, you could potentially be overwriting the ThreadX memory pool.

The NewLib heap and the ThreadX allocation are working completely unaware one of an other. If it is up to the user to be aware of that, that should be documented, it can cause quite a lot of problems. Otherwise, are you planing to offer a better integration?

Looking forward to reading your thoughts.

Thanks.

ALABSTM, 2021-01-15:
Hi @bytesontherocks,

Thank you for this contribution. Your report will be forwarded to our development teams. I will get back to you as soon as they provide me with their feedback.

With regards,

ALABSTM, 2021-05-27:
Hi @bytesontherocks,

Our development teams confirmed your point. However, they said good management of allocations in both `NewLib` heap and `ThreadX` memory pool falls on the "shoulders" of the user application.

From our side, we will try to highlight such an important point in our documentation.

Thank you again for having reported this point.

With regards,

closed 2021-08-02 by bytesontherocks