STMicroelectronics/x-cube-azrtos-h7 #3: Erratic behavior of H735 WebServer example

BrunoCoppi, 2021-02-15:
Hello,
I am porting the examples to a different H7 microcontroller; I have found and solved a couple of Data Cache related issues, causing corrupted data transfers in the H735 WebServer example.

Microcontroller used: STM32H743ZI on custom board

TWO Toolchains used with same behavior
Toolchain: KEIL MDK-ARM v.5.33 - Arm Compiler v.6.15
Toolchain: STM32CubeIDE v.1.5.0

Effect: On my board, I have an SD-Card containing the provided example files for web server; the PC browser receives corrupted images and corrupted text. The behavior is erratic.

My analysis led to the following results:
The Data Cache is not correctly handled; some memory buffers are NOT 32 bytes aligned and this leads to memory corruption.

Proposed modifications:

1. The file **nx_user.h** should be modified to the proper packet alignement constant
   line 182
       #define NX_PACKET_ALIGNMENT 	  				(32)

2. The SD-Card driver file **fx_stm32_sd_driver.c** should be modified to ensure proper scratch buffer use when needed
   line 122
         if ((ULONG)(media_ptr->fx_media_driver_buffer) & 0x1F)
   line 149
         if ((ULONG)(media_ptr->fx_media_driver_buffer) & 0x1F)

3. The SD-Card driver file **fx_stm32_sd_driver.c** should be modified to remove the un-needed memset function call (that causes other collateral effects on Data Cache)
   line 325 - **REMOVE**
        memset(scratch, '\0', DEFAULT_SECTOR_SIZE);

The mentioned modifications solved the issue on my board.
Thank you for your kind attention, best regards
Bruno

ALABSTM, 2021-02-15:
Hi @BrunoCoppi,

Thank you for this report and for your fix proposals. It will be forwarded to our development teams. Thank you once more.

With regards,

ALABSTM, 2021-05-28:
Hi @BrunoCoppi,

According to our development teams, the SD/DMA requires buffers that are **4-byte** aligned. Hence, there is no need to have a 32-byte alignment.

For a proper operation of the examples, the `NX_PACKET` should be allocated in a memory region that is **non-cacheable**. This can be achieved using the **MPU**.

With regards,

ALABSTM, 2021-06-17:
Hi @BrunoCoppi,

I hope you are fine. I also hope the above answer is helpful. Please allow me to close this issue now.

With regards,

closed 2021-06-17 by ALABSTM

BrunoCoppi, 2021-06-17:
Thank you Ali and please forgive my late reply.
Yes, your team just chose a different way to deal with the issue.
We preferred to keep the cache on data memory and using a different 
workaround.
They are just two different approaches...
I am fine with your answer.
Thank you for your support, best regards
Bruno

-- 

Tecnologix logo

-- 
Bruno Coppi

TECNOLOGIX s.r.l.
Via dei Biancospini 6
I 20146 Milano Italy
Tel    +39 02 48954230
Fax    +39 02 471106
e-mail ***@***.***
web    http://www.tecnologix.it

Le informazioni contenute nella presente comunicazione e i relativi allegati possono essere riservate e sono, comunque, destinate esclusivamente alle persone o alla Società sopraindicati e non sono da considerarsi comunicazioni personali, quindi eventuali risposte potranno essere conosciute da persone appartenenti all’azienda.

The information in this communication (which includes any documents with it) is confidential and meant only for above-mentioned persons and/or Companies; the information is not personal therefore, it is intended for the addressee only and any replies will be known by people belonging to the company.