STMicroelectronics/x-cube-azrtos-h7 #4: Port Ux_Device_HID_CDC_ACM  failure

geek-slt, 2021-04-16:
**Caution**

The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post your report to the **ST Community** in the STM32 MCUs dedicated [page](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**

 * The board (either ST RPN reference or your custom board)
 * IDE or at least the compiler and its version

**Describe the bug (skip if none)**

A clear and concise description of what the bug is.

**How to reproduce the bug (skip if none)**

1. Indicate the global behavior of your application project
2. List the modules that you suspect to be the cause of the problem (Drivers, BSP, MW...)
3. Describe the use case that generates the problem
4. How we can reproduce the problem

**Additional context**

If you have a first analysis, an enhancement, a fix or a patch, thank you to share your proposal.

**Screenshots**

If applicable, add screenshots to help explain your problem.
Toolchain:  MDK-ARM  Version:5.34.00
Mcu: STM32H743XIHb
Question:
1.I refer  x-cube-azrtos-h7-main->STM32H747I-DISCO->App->USBX->Ux_Device_HID_CDC_ACM for porting USBX and ThreadX
2.ThreadX Working fine
3.MX_USB_Device_Init() Initialization successful
4.No new devices appear in Windows Device Manager
5.Use MDK for Debug MX_USB_Device_Init() can be executed completely, but no USB appears. How can I solve it?

ALABSTM, 2021-04-16:
Hi @geek-slt,

Thank you for this report. It will be forwarded to our development teams for analysis. We will get back to you as soon as we have an answer.

With regards,

arilink, 2021-04-19:
@ALABSTM   very much looking forward to your help

arilink, 2021-04-19:
I have some new developments and need to sync with you
1.The previous USB did not reflect anything because the USB in the example was initialized as an HS device, but on my hardware it was an FS device. When I changed the USB initialization part to an FS device, the device manager prompts that a new device is inserted, but Failed to get device descriptor
2.I tried a series of operations such as changing the descriptor, but the problem still exists, win10 always prompts that the device descriptor request failed
3.How can I continue troubleshooting?

Mathullah, 2021-04-19:
@arilink I have similar issues in the past with another USB library on an STM32F7 device. Try the following:
- Disable Data Cache on the MCU (if enabled)
- Disable USB DMA transfer (if enabled)
- Configure the MMU 
Hope this helps.

arilink, 2021-04-20:
1. I try to close I DCache and it doesn't work
2. I did not enable USB DMA
3. No MMU

Can I send the project to you to check？

ALABSTM, 2021-04-20:
Hi @arilink,

Yes please do. This would be helpful.

With regards,

arilink, 2021-04-20:
@ALABSTM Can you leave your mailbox? I send the project to you

ALABSTM, 2021-04-21:
Hi,

@arilink, you can send your project attached to you comment on GitHub. Remove any generated or unnecessary file to optimize the size, zip it then attached to your comment as said. [Here](https://docs.github.com/en/github/managing-your-work-on-github/file-attachments-on-issues-and-pull-requests) is how to proceed.

@Mathullah, thank you very much for your contribution and support.

With regards,

arilink, 2021-04-26:
[Threadx - HID+CDC.zip](https://github.com/STMicroelectronics/x-cube-azrtos-h7/files/6375948/Threadx.-.HID%2BCDC.zip)

arilink, 2021-04-26:
@ALABSTM I have a problem with my computer and I didn’t reply in time. I have uploaded the complete MDK project. Please check it out.

ALABSTM, 2021-04-28:
Hi @arilink,

Your project files have been well received. We are investigating the behavior you described. I will get back to you as soon as I have feedback.

With regards,

arilink, 2021-05-05:
@ALABSTM how's the progress？

ALABSTM, 2021-05-05:
Hi @arilink,

Still working on it.

With regards,

ALABSTM, 2021-05-24:
Hi @arilink,

We noticed the structure and the generated code of your project are different from the default ones generated by Cube MX (**e.g.**, implementation of the `SystemClock_Config()`, `bsp.c` file added, `MX_GPIO_Init()` and `MX_DMA_Init()` functions moved to `bsp.c`). Would you mind regenerating your project using latest version of Cube MX and **send us the IOC file** please? Thank you in advance.

Besides, we noticed the following:
* In the file `./bsp/bsp.c`, memory region should be left `MPU_ACCESS_NOT_BUFFERABLE` and `MPU_ACCESS_NOT_CACHEABLE` as recommended by our development teams (**lines 290** and **291**).
* The provided application constituting your starting point uses the **high** speed (HS) while your project uses the **full** speed (FS) one.
  * In the file `./bsp/src/usb_otg.c`, value `GPIO_AF10_OTG1_FS` shall be replaced by `GPIO_AF10_OTG2_FS` at **line 103** (this should be automatically fixed once the project **regenerated with Cube MX**).
  * In the file `./usbx_app/app_usbx_device.c`, **line 360** needs to be updated as below:
```diff
  /* initialize the device controller driver*/
-  ux_dcd_stm32_initialize((ULONG)USB_OTG_HS, (ULONG)&hpcd_USB_OTG_FS);
+  ux_dcd_stm32_initialize((ULONG)USB_OTG_FS, (ULONG)&hpcd_USB_OTG_FS);
```
* The provided application constituting your starting point **does not support** the ARM Compiler v6.
  * However, your project configuration indicates this version is the one used.
  * Please pay attention that folder `ac6` is **not related** to ARM Compiler v6 (as the resemblance with folder name `ac5` may suggest) but rather to [AC6-Tools](https://www.ac6-tools.com/).
* To be able to use the ARM Compiler v6, the update below needs to be made to the following files:
  * `./bsp/src/usb_otg.c`
  * `./usbx_app/app_usbx.c`
  * `./usbx_app/ux_device_cdc_acm.c`
```diff
-#elif defined ( __CC_ARM ) /* MDK ARM Compiler */
+#elif defined ( __CC_ARM ) || defined ((__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050)) /* MDK ARM Compiler */
```
* Once the above update made, please do not forget to align the section names in your scatter file (**e.g.**, `.UsbxAppSection`).

We hope this helps. We will wait your IOC file to further investigate.

With regards,

ALABSTM, 2021-05-25:
Hi @arilink,

Would it be possible to have the trace of the USB enumeration phase once your device is plugged? This would also be of great help.

Thanks,

arilink, 2021-05-31:
@ALABSTM  I will try it later, thank you very much for your help

arilink, 2021-06-29:
@ALABSTM  I used CubeMX to configure ThreadX and regenerated the code. I copied the USBX/APP with reference to the example, but there were many errors. Can you check it out for me?

arilink, 2021-07-01:
Hello, I used the cubemx configuration and added ThreadX and USBX related components in the Software Packs, and added the USBX sample program imitating the H7 example, but the USB still does not work. Can you help me check the problem? I have been troubled by USBX for a long time.

arilink, 2021-07-05:
@ALABSTM Thank you very much for your support, the problem has been solved

closed 2021-07-05 by arilink

Mathullah, 2021-07-05:
@arilink Can you short explain how you solved this? The community would appreciate it very much, especially myself since we're planing to start a new project with a similar configuration. Thanks!

arilink, 2021-07-05:
1. Check and confirm whether the USB used in the project is FS or HS and adjust the USB initialization part of the code in app_usbx_device.c->MX_USB_Device_Init
2. Check and confirm that the USB interrupt is turned on
3. Check and confirm that the USB clock is correctly configured
4. Check and confirm that the size of the memory pool allocated to the USB task in the tx_application_define() function is appropriate
5. If none of the above is a problem, check the USB App program

Mathullah, 2021-07-07:
@arilink thank you!