STMicroelectronics/x-cube-azrtos-h7 #5: About USB endpoint FIFO read and write issues

arilink, 2021-07-07:
I have successfully ported ThreadX and USBX on my own board and successfully run the USB CDC+HID example
But when I tried to understand the USB workflow, I ran into a problem. I can only roughly understand the need to prepare device descriptors, report descriptors, etc. But I don’t know the USB FIFO endpoints and the link between reading and writing. The related code is very complicated. Whether to provide A note to explain this?

CHAMSTM, 2021-07-07:
Hello @arilink ,
Could you please have a look on this wiki page and let us know if it would be helpful to progress on your implementation?
https://wiki.st.com/stm32mcu/wiki/Category:USB

And then for the FiFo configuration, normally you don't need to go much into details since the HAL provides you the abstraction layer for configuration, reading and writing. You just need to call the upper layer (class) functions.
In case you need deep insight on FIFOs configuration and usage, you can refer to the STM32 reference manual that contains all these deatails.

Kind Regards,

arilink, 2021-07-08:
@CHAMSTM hello，
It’s hard for me to convince myself to develop with questions. Although the system works, I don’t understand why.
0. This document and Azure USBX document https://docs.microsoft.com/zh-cn/azure/rtos/usbx/ I have studied carefully but I can’t solve all my doubts
The example I refer to is x-cube-azrtos-h7-main\x-cube-azrtos-h7-main\Projects\STM32H747I-DISCO\Applications\USBX\Ux_Device_HID_CDC_ACM\MDK-ARM
I will list all my doubts below and ask you to help me answer them.
(1) tx_application_define()->App_USBX_Device_Init()->usbx_app_thread_entry()->MX_USB_Device_Init() calls HAL_PCDEx_SetRxFiFo() and 4 HAL_PCDEx_SetTxFiFo() in the initialization function of the USB hardware
(2) App_USBX_Device_Init()->USBD_Get_Device_Framework_Speed() is for generating descriptors. I used USBLyzer for analysis. In the end, there is 1 configuration descriptor, 3 interface descriptors, 4 endpoint descriptors, and 1 report descriptor.
my question:
1. How does the FIFO created in MX_USB_Device_Init() match the endpoint?
2. There are 4 TxFiFo created in MX_USB_Device_Init(), but in fact there are only 3 IN endpoints. Why?
3. In the USBX document of Azure, the descriptor is directly made into an array, while the sample program uses a lot of functions to implement it. What are the benefits of this? Not easy to analyze
4.ux_app_parameters_change() corresponds to CDC Control. How can I operate CDC Control to change the baud rate and other operations?
5. ux_device_class_cdc_acm_read() reads data. How to make sure it reads the correct endpoint data?
[USBX Study.zip](https://github.com/STMicroelectronics/x-cube-azrtos-h7/files/6780912/USBX.Study.zip)

arilink, 2021-07-08:
[USBX Study.zip](https://github.com/STMicroelectronics/x-cube-azrtos-h7/files/6780912/USBX.Study.zip)

arilink, 2021-07-22:
@CHAMSTM Did no one answer the question?

closed 2021-08-02 by arilink