STMicroelectronics/x-cube-azrtos-h7 #8: About FileX using Dcache and DMA

zi-niu, 2021-08-05:
Hello,
I have successfully ported ThreadX and FileX on my own board STM32H735G DK and successfully run the Fx_uSD_File_Edit example.
But when I enable the FX_ENABLE_EXFAT macro and use the exfat format SD card, the program makes an error when opening the SD device and returns FX_IO_ERROR. When I disable DCache, the program runs normally.
So I checked the code and found that there seems to be some problems in the processing of DCache, 
in x-cube-azrtos-h7/Middlewares/ST/filex/common/drivers/fx_stm32_sd_driver.c line 122, 
`unaligned_buffer = (UINT)(media_ptr-> fx_media_driver_buffer) & 0x3;`
This is used to determine whether the four bytes are aligned, and if so, use DMA transmission directly. In line 300 and line 336, SCB_InvalidateDCache_by_Addr() is used to maintain data consistency, but SCB_InvalidateDCache_by_Addr() requires 32-byte data alignment.
I changed the code to
`unaligned_buffer = (UINT)(media_ptr->fx_media_driver_buffer) & 0x1f; `
The program seems to be able to use exfat and Dcache at the same time.
I want to know if I am doing this right? And why there is no problem when using the FAT32 file system. 

Thank you .

ALABSTM, 2021-08-17:
Hi @zi-niu,

Thank you very much for this clear and detailed report. It has been already forwarded to our development teams. We will get back to you as soon as we have their answer.

With regards,

arilink, 2021-08-25:
@zi-niu Did you configure and generate the project through CubeMX and then port ThreadX and FileX? I also want to perform a similar operation, but I am not sure about the SD card DMA configuration. The official example is not easy to see the configuration process. Can you refer to your project?

arilink, 2021-08-28:
Are you using the same chip and engineering directly? I’m using H750 and can’t directly use H735’s project. Porting H735’s code keeps prompting that the mount fails.

arilink, 2021-08-28:
Hi zi-niu
Are you Chinese? I feel that you have a deep research on ThreadX and its supporting components. I want to learn more from you.

***@***.***
 
From: zi-niu
Date: 2021-08-27 18:00
To: STMicroelectronics/x-cube-azrtos-h7
CC: arilink; Mention
Subject: Re: [STMicroelectronics/x-cube-azrtos-h7] About FileX using Dcache and DMA (#8)
@arilink https://github.com/zi-niu/STM32H735-FileX-demo

zi-niu, 2021-08-30:
@arilink Yes, I am Chinese. You can friend me on WeChat: Calf_K