STMicroelectronics/x-cube-azrtos-h7 #10: Does USBX MSC support Nor FLASH?

arilink, 2021-09-09:
SD MSC has been successfully run on USBX and can read and write normally. Can Nor FLASH also achieve the same function?

CHAMSTM, 2021-09-28:
Hi @arilink ,

Actually, USB MSC interface is agnostic of the low layer medium used.

So you can use SD or NOR or any type of memory  (as long as it supports direct read/write operations) by replacing the calls to SD driver callbacks by calls to the new memory callbacks.

But it is important to pay attention to potential specific features of a memory (ie. dual bank, addressing scheme, ...) that may cause issues: USB MSC will simply expose the memory to the Host and let him manage it directly through read/write. So Host expects to have direct access to the address locations and will not perform additional operations (ie. setup bank, add offsets, manage sectors..).

Of course memories like NAND could not be interfaced to USB MSC directly, since they require specific management layer.
For the NOR I think there should be no major issues since BSP/HAL drivers would take in charge init and read/write operations.

We don't have such implementation today and it is not planned, but I think it is possible to implement with some reasonable effort.

I hope this helps.

Kind Regards,

arilink, 2021-09-30:
ok i will try it  
thankyou very much

arilink, 2021-09-30:
@CHAMSTM 
AzureRTOS and its components are very good. I have been working hard to apply it to actual projects. Our projects mainly use ThreadX and USBX. The example you gave is to use H7 step by step down to F4. I put it on F1 after testing on H7. However, I encountered a lot of problems in this process. I tested HID IN+CDC on H7 at the same time. The transmission rate of HID can reach 1000 times per second, but the same program cannot work normally on F1. Can you help? I submitted a new ISSUE https://github.com/azure-rtos/usbx/issues/39

Bharath-Zion, 2024-08-27:
Hi @arilink-tech 
I'm working on STM32U0 series MCU and SPI NOR Flash "W25Q32JV".
I'm not able to implement the MSC using USBX. Can you help me to do that?