STMicroelectronics/x-cube-azrtos-h7 #14: HTTPS server 'hello world' example

hmckinlay, 2021-11-01:
Environment: Windows based CubeIDE/CubeMX and a Nucleo-H753ZI

Is someone able to post a working 'hello world' web page example using the HTTPS server? One that works with Chrome and Firefox? 

Right now I am getting:
[image]

I have tried several combinations of cipher suites with no luck. Most of the config is from this example:
https://community.st.com/s/question/0D53W00000nZEvLSAW/http-to-https-web-server-on-stm32h735gdk-based-on-examples

CubeMX Crypto Core Settings:
[image]

CubeMX TLS Core settings:
[image]

Other references:
https://github.com/STMicroelectronics/x-cube-azrtos-h7/blob/main/Middlewares/ST/netxduo/samples/demo_netxduo_https.c
https://docs.microsoft.com/en-us/azure/rtos/netx-duo/netx-secure-tls/chapter2
https://docs.microsoft.com/en-us/azure/rtos/netx-duo/netx-duo-web-http/chapter2
https://github.com/STMicroelectronics/x-cube-azrtos-h7/tree/main/Projects/STM32H735G-DK/Applications/NetXDuo/Nx_WebServer

I now have nearly have working the basis of a functional system:
- HTTP server 
- redirect from the base of the web server to /index.html
- serving multiple files html/css/js 
- serving large pages 
- injecting an uptime server side variable into the page
- AJAX to dynamically update the uptime
- posting a new uptime back from the browser  
- basic authentication

In my opinion this stuff is far more useful as an examples, rather than the http(s) server sending a packet to itself.

hmckinlay, 2021-11-28:
Hello? Anyone?

hmckinlay, 2021-12-16:
Looks like the https server doesn't work with Chrome or Firefox 

https://github.com/azure-rtos/netxduo/issues/67

ALABSTM, 2022-01-05:
Hi @hmckinlay,

Excuse this delayed reply. Our development teams are already having a look to what you reported. We will get back to you as soon as they provide us with their feedback.

With regards,