STMicroelectronics/x-cube-azrtos-h7 #15: Update info on README.md

escherstair, 2021-11-12:
README.md has info on the supported toolchains.
The name [RealView Microcontrollor Development Kit](https://github.com/STMicroelectronics/x-cube-azrtos-h7/blob/a455a40ad5213625b66810295789cb268cbc50a3/README.md?plain=1#L31) is an old one.
I suggest to replace it with the up-to-date name "Keil MDK"

ALABSTM, 2021-12-07:
Hi @escherstair,

Thank you for having reported. Your point will be forwarded to our development teams.

With regards,

ALABSTM, 2022-02-09:
Hi @escherstair,

According to our development team, your point is relevant yet minor. As the update, if carried out, shall be done on all X-Cube Azure RTOS packages, choice has been made to leave the toolchain's name unchanged.

Please allow me to close this issue. Thank you for your comprehension and thank you again for having reported.

With regards,

closed 2022-02-09 by ALABSTM