STMicroelectronics/x-cube-azrtos-h7 #16: Azure IOT Addon

hmckinlay, 2021-11-29:
Hardware: Nucleo-H753ZI
Development Environment: Windows based STM32CubeIDE 1.8.0

It would be great if this library / environment would include the Azure IoT addon, seeming like a strange omission. 
https://github.com/azure-rtos/netxduo/tree/master/addons/azure_iot

hmckinlay, 2021-12-01:
For anyone else working on this, you can do this, not too bad when you know how. It will be nice when it is included in this library.

You need the addon from here:
https://github.com/azure-rtos/netxduo/tree/master/addons/azure_iot

It has a bunch of references in here so you'll need to grab this one: 
https://github.com/Azure/azure-sdk-for-c

I now have working the simplest example in here, https://github.com/azure-rtos/netxduo/tree/master/addons/azure_iot/samples the one using 'IoTHub connect using Symmetric key'.

The only trick was the time the SNTP addon gets the time in NTP Epoch, this needs to be converted to UNIX time. Bit of a shame this isn't done automagically.

ALABSTM, 2021-12-07:
Hi @hmckinlay,

Thank you for having shared what you were able to do so far. I will let our development teams know about it.

With regards,

CHAMSTM, 2022-01-05:
Hi @hmckinlay , could you please have a look on this [repo](https://github.com/STMicroelectronics/Azure-cloud-samples) and check if it fits your needs ?
Azure IoT is not part of the X-CUBE-AZRTOS-XX packages but it is provided in a dedicated package on selected series.
I hope it helps.

hmckinlay, 2022-01-17:
> Hi @hmckinlay , could you please have a look on this [repo](https://github.com/STMicroelectronics/Azure-cloud-samples) and check if it fits your needs ? Azure IoT is not part of the X-CUBE-AZRTOS-XX packages but it is provided in a dedicated package on selected series. I hope it helps.

That lib looks similar to what I have done, thanks. Is the Azure IoT addon on the roadmap to be a part of X-CUBE-AZRTOS-XX?
Cheers,
Hamish.