STMicroelectronics/x-cube-azrtos-h7 #17: Nx_TCP_Echo_Client example on STM32H735G-DK

escherstair, 2021-11-29:
This is an enhancement request.
Nx_TCP_Echo_Client example is available for NUCLEO-H723ZG.

Is it possible to add this example for STM32H735G-DK too?

Thanks

rahmanih, 2021-11-29:
Hi @escherstair,
you can easily port it without any issue:
- create a copy of the "NX_UDP_Echo_Client" that is running on the STM32H735G-DK
- use the files "app_netxduo.h" & "app_netxduo.c" from the "NX_TCP_Echo_Client" to override the ones of the "NX_UDP_Echo_Client"
 
could you try and let us know?

escherstair, 2021-11-30:
Hi @rahmanih,
I followed your suggestion.
Everything works fine, with an additional step:
- inside "app_netxduo.h" replace `#include "stm32h7xx_nucleo.h"` with #include `"stm32h735g_discovery.h"`
Then I changed a little bit some file names and references from NX_UDP_Echo_Client to NX_TCP_Echo_Client

rahmanih, 2021-11-30:
well done!

ALABSTM, 2021-12-07:
Hi,

@rahmanih, thank you for your support. @escherstair, glad to read you could port this example to your DK board. Please allow me to close this issue. 

With regards,

closed 2021-12-07 by ALABSTM