STMicroelectronics/x-cube-azrtos-h7 #20: How to use Threadx

zsjlovestudy, 2022-01-27:
Environment:
STM32CubeIDE 1.8.0
STM32CubeMX 6.4.0
x-cube-azrtos-h7 2.0.0

I only initialized the RCC clock and an LED IO port，then i choose threadx core in software pack of x-cube-azrtos-h7.
After I added an LED flashing thread, but it didn't run
[image]

c4chris, 2022-01-27:
You need to allocate memory before calling tx_thread_create, e.g.

```
CHAR *pointer;

ret = tx_byte_allocate(byte_pool, (VOID **) &pointer, DEFAULT_STACK_SIZE, TX_NO_WAIT);

ret = tx_thread_create(&my_thread, "My Thread", my_thread_entry, 0, pointer, DEFAULT_STACK_SIZE, DEFAULT_THREAD_PRIO,
                  DEFAULT_PREEMPTION_THRESHOLD, TX_NO_TIME_SLICE, TX_AUTO_START);
```

You might want to check the value of `ret` after each call

zsjlovestudy, 2022-01-27:
> CHAR *pointer;
> 
> ret = tx_byte_allocate(byte_pool, (VOID **) &pointer, DEFAULT_STACK_SIZE, TX_NO_WAIT);

thank you very much！ I was careless and tried Azure for the first time.

zsjlovestudy, 2022-01-28:
> You need to allocate memory before calling tx_thread_create, e.g.
>
> ```
> CHAR *pointer;
>
> ret = tx_byte_allocate(byte_pool, (VOID **) &pointer, DEFAULT_STACK_SIZE, TX_NO_WAIT);
>
> ret = [...]

May I ask you another question, if I choose dynamic memory allocation, how should I allocate for it?  

[image]
[image]