STMicroelectronics/x-cube-azrtos-h7 #24: FX_IO_ERROR on Fx_NoR_Write_Read_File example

bdlr2, 2022-05-03:
**Describe the set-up**

 * The board: STM32H735G-DK
 * IDE: IAR 9.10.2
 * Project: Fx_NoR_Write_Read_File example

**Describe the bug (skip if none)**

I start the example out of the box, generated with CubeMX for STM32H735G-DK. The `fx_media_format` at line 147 in app_filex.c returns a FX_IO_ERROR, which ends the program.

**How to reproduce the bug (skip if none)**

Start the app and run it step by step up to the `fx_media_format` call in app_filex, you'll see the return value that is not FX_SUCCESS as it should.

ALABSTM, 2022-07-29:
Hi @bdlr2,

Your report has been forwarded to our development teams. They will be back to you after a deeper analysis of the issue you reported.

With regards,

**PS:** Please excuse the delay it may take to reply.

rahmanih, 2022-07-29:
Hi  @bdlr2,
Did the NOR memory contain any data before running the application?

regards
Haithem.

bdlr2, 2022-08-08:
HI @rahmanih,

I finally found that it's indeed because of the NOR that was not completely erased that this problem happened.
I agree it's not a bug, but may I suggest to add some info about this prerequisite in the documentation, or to perform a test to erase the NOR automatically ?

Best regards

rahmanih, 2022-08-08:
Hi  @bdlr2 
Great news!
Actually, the request to erase the memory is mentioned in the [README.md](https://github.com/STMicroelectronics/x-cube-azrtos-h7/tree/main/Projects/STM32H735G-DK/Applications/FileX/Fx_NoR_Write_Read_File#readme)
```
The NOR flash should be erased prior to format either by the application or by the STM32CubeProgrammer, this allows LevelX and FileX to create a clean FAT FileSystem.

Chip erase operation takes considerable time when done by the application, therefore it is disabled by default.

To enable it, please define the flag LX_STM32_OSPI_ERASE to 1 in "lx_stm32_ospi_driver.h":
```