STMicroelectronics/x-cube-azrtos-h7 #26: usb dma

chengshuihang, 2022-06-26:
您好！我想启用USB的DMA功能，我看到USBX中使用_ux_utility_memory_allocate来保证内存对齐，如果我需要启用cache，是不是只需要在HAL_HCD_HC_SubmitRequest前后调用SCB_CleanDCache_by_Addr或者SCB_InvalidateDCache_by_Addr函数即可？如下面截图这样的修改。
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ALABSTM, 2022-07-27:
Hi @chengshuihang,

Would you mind resubmitting this issue in English please? In the meanwhile, please allow me to close this one as written in Chinese.

With regards,

closed 2022-07-27 by ALABSTM

chengshuihang, 2022-10-11:
您关闭它吧，我不介意。另外我英文水平可能不够，就不再用英文提交了 

流利偷。。。
***@***.***