STMicroelectronics/x-cube-azrtos-h7 #41: Double Free Issue

testfuna20230406, 2025-04-25:
#### Overview  
A double free may occur in the STM32 USBX host controller code when `ux_stm32_ed_setup` and `ux_stm32_ed_data` point to the same memory.

---

#### Step 1: Memory freed in control transfer  
In `ux_hcd_stm32_request_control_transfer()` (file: `ux_hcd_stm32_request_control_transfer.c`, around line 239), the following code frees `ed->ux_stm32_ed_setup`:

```c
_ux_utility_memory_free(ed -> ux_stm32_ed_setup);
ed -> ux_stm32_ed_setup = UX_NULL;
```

---

#### Step 2: Memory freed again in interrupt  
Later, `_ux_hcd_stm32_request_trans_finish()` (in `ux_hcd_stm32_request_trans_finish.c`) is called in interrupt context.  
It contains this code:

```c
if (ed -> ux_stm32_ed_data != ed -> ux_stm32_ed_setup)
{
    _ux_utility_memory_free(ed -> ux_stm32_ed_data);
    ed -> ux_stm32_ed_data = UX_NULL;
}
```

However, if `ed->ux_stm32_ed_data` and `ed->ux_stm32_ed_setup` were originally the same address (e.g., `0x12345678`), the same memory is freed twice:
1. Once in `request_control_transfer`
2. Once again in `request_trans_finish`

This causes a **double free**, which may lead to memory corruption or crash.

ALABSTM, 2025-04-28:
Hi @testfuna20230406,

Thank you for this report. Did you actually experienced crashes with some real application or is this rather a deduction you came to through code review mainly?

With regards,

testfuna20230406, 2025-04-30:
Hi @ALABSTM 

I did not see any real crashes, but the callback function I registered using ux_utility_error_callback_register() returned UX_MEMORY_CORRUPTED.
Also, the function _ux_utility_memory_free() calls _ux_system_mutex_on(), but I think mutexes should not be used in an interrupt context.
Because of this mutex problem and the memory corruption error, I started checking the code.
That is how I found this issue and decided to report it.

Regards,